GeekDetour/BrickLayers #2: Linux - PrusaSlicer bug

rodrykthingol, 2025-03-17:
First off thank you! 

Second, here's the bug. Ran the script on prusaslicer and got this error:

[image]

Something about missing the pypy3. So i did it manually in the terminal.

[image]

Now it gives me this error:
Traceback (most recent call last):
  File "/home/thingol/Apps/Bricklayers_GD/bricklayers.py", line 2174, in <module>
    outfile.write(processed_line)
TypeError: unicode argument expected, got 'NoneType'

GeekDetour, 2025-03-17:
Hi Rodrigo! Can you include the STL that was generated? Were you using Classic Engine or Arachne?

jungletek, 2025-03-17:
I managed to work out a fix and got it to generate a full modified gcode file too.

Line 1788:
`
                if not self.yield_objects:
                    yield from (gcodeline.to_gcode() for gcodeline in buffer_lines)
                else:
                    yield from buffer_lines
                buffer_lines.clear()
`
Becomes:
`
                if not self.yield_objects:
                    yield from (gcodeline.to_gcode() for gcodeline in buffer_lines if gcodeline is not None)
                else:
                    yield from (gcodeline for gcodeline in buffer_lines if gcodeline is not None)
                buffer_lines.clear()
`
and lines 1623 and 1805 go from:
`
                    buffer_lines.append(myline)
`
to:
`
                    if myline is not None:
                        buffer_lines.append(myline)
`
 Additionally, I slightly modified the initialization of `self.gcode` so it reads `self.gcode = gcode or "" ` for safety.

Sorry I didn't make a proper pull request, but I've been fiddling with this for a couple hours (incl. configuring PrusaSlicer for this and Klipper) after being up all night and being excited to try this out after seeing the video.

GeekDetour, 2025-03-17:
Jungletek, oh man... I appreciate the time you fought it.  But, first, take a look at the test script:
https://github.com/GeekDetour/BrickLayers/blob/main/sample_tests/simpletest.sh

Please, In your terminal:
Go inside the "sample_tests" folder
chmod +x simpletest.sh
./simpletest.sh

You should see it running correctly without any modification.

jungletek, 2025-03-17:
I'm pretty fried, so I'm not sure I follow. Yes that works (once I translate it to DOS syntax, don't let the prompt fool you, that's a CMD prompt), but we don't want to only process the test file, we want to process arbitrary sliced gCode, and more importantly, do it through the slicer.
If I don't make the modifications I posted, I get the same error in PrusaSlicer that the OP got, `line 2174, in <module> outfile.write(processed_line) TypeError: write() argument must be str, not None`, just like I get it when I invoke the script manually from the command line.

Just now, I tested the modified script in PrusaSlicer and for the first time, was able to get the script to successfully run on an arbitrary sliced file, and checked that the layers look as expected. Isn't that the point? What am I missing, brother?

GeekDetour, 2025-03-17:
Oh, you are not on linux or Mac. Let me upload a .bat version... Give me a few moments.

GeekDetour, 2025-03-17:
By the way, please forgive my communication problems now. I've been working on this program and the video for a few weeks now, and sleeping very little, ha ha.

jungletek, 2025-03-17:
OP's error:
> Now it gives me this error: Traceback (most recent call last): File "/home/thingol/Apps/Bricklayers_GD/bricklayers.py", line 2174, in outfile.write(processed_line) TypeError: unicode argument expected, got 'NoneType'

My error:
> `line 2174, in <module> outfile.write(processed_line) TypeError: write() argument must be str, not None`

There was an error in the script, as far as we both could tell. The function was expecting a string as an input, but it was choking when it got to type "None". I added the logic to handle "None" and made the script work as expected. Now it doesn't throw errors in the console or when invoked through PrusaSlicer, and works as intended, and I'm happy for that.

And it's a very cool script, thank you for your work!

I'm guessing this is one of those "it works on my dev environment but breaks when I send it to other people" sort of things perhaps? Maybe some sort of cross-platform difference that works in a technically undefined state on one platform (Linux for example), and then poops itself on Windows because who knows?

> By the way, please forgive my communication problems now. I've been working on this program and the video for a few weeks now, and sleeping very little, ha ha.

Not a problem at all, I think we may all be burning the candle at both ends ;)

jungletek, 2025-03-17:
Oh, if it helps you, I can provide a super simple input gcode file (just a cube) and the output of your unmodified script. The script was crashing about 30% of the way through, so examining the last line(s) written to the incomplete file may give you clues as to what exactly made your script choke for us, but not for you in your tests.

GeekDetour, 2025-03-17:
Yes, it would be great because I need to understand which circumstance generate a None line. It would be lovely. ❤️

jungletek, 2025-03-17:
Okay, interesting results.

Generating a cube primitive via PrusaSlicer's right-click menu and slicing: your unmodified script SUCCESSfully processes the gcode and the object has brick layers as expected.

Dropping the BrickLayersChallengeSimple.stl (the model I happened to first test the script functionality with) on the build plate & slicing: your unmodified script FAILS part-way through.

This is the untouched, freshly-sliced object:
[BrickLayersChallengeSimple_0.20mm_15%-infill.txt](https://github.com/user-attachments/files/19297191/BrickLayersChallengeSimple_0.20mm_15.-infill.txt)

This is the partially-processed result of the script crashing midway:
[BrickLayersChallengeSimple_0.20mm_15%-infill_brick.txt](https://github.com/user-attachments/files/19297193/BrickLayersChallengeSimple_0.20mm_15.-infill_brick.txt)

rodrykthingol, 2025-03-17:
Ok. So finally managed to get it to work...

[image]

Running PrusaSlicer 2.9.1 - on Fedora 41 (Linux)

Heres my config file, configured for a voron 0.2:
(remove the .txt)

[config.ini.txt](https://github.com/user-attachments/files/19298733/config.ini.txt)

The stl and gcode output:

[3dbenchy.stl.txt](https://github.com/user-attachments/files/19298735/3dbenchy.stl.txt)

[3dbenchy.gcode.txt](https://github.com/user-attachments/files/19298736/3dbenchy.gcode.txt)

The main culprits for me that resulted in errors and artifacting are this:

Number one:
"/usr/bin/python" "/home/thingol/Apps/Bricklayers_GD/bricklayers.py" -startAtLayer 3 -extrusionMultiplier 1.05 -enabled 1;

Use python, instead of pypy. Don't know why it doesn't work.

Second:
Turn wipe on in the printer settings

[image]

For whatever reason if thats turned off it prevents from even running the bricklayer script

Three:
Turn scarf joint off.

[image]

Four:
This little bastard removes all the remaining artifacts. So turn detect thin lines on.

[image]

rodrykthingol, 2025-03-17:
Oh also forgot, you can turn the external perimeters first off too. Just had it on by default due to experience with tentech's script.

rodrykthingol, 2025-03-17:
Will try testing this in a couple of different models. But the results seem encouraging.

GeekDetour, 2025-03-17:
Wow! That was amazing. Thank you so much!

I need to check why unmarking "wipe while retracting" is breaking things. About the wall order, the optimal setting is inner/outer.

GeekDetour, 2025-03-17:
After your testings, can you see that the Perimeters are indeed Bricked? Can you still see they alternating? (just checking :)

rodrykthingol, 2025-03-17:
Did a quick run.

Benchy confirmed.

[image]

Octopus confirmed.

[image]

Engine Block v6 confirmed. But strangely enough there are some artifacts.

[image]

GeekDetour, 2025-03-17:
You should print a small piece that has those artifacts. You might see they are not actually printed. It's just from the preview. I spent quite some time also making sure I set the right ;WIDTH and ;HEIGHT after I moved the walls to the "bricked" layer, to not interfere with what comes right after. Those things are not printed... but they shape how the preview renders the lines 🥵

jungletek, 2025-03-17:
> I have your change as a "wildcard" in my hand. I want to understand the underlying reason that is causing the problem in Windows, without having to resort to an "if" to skip lines if None. I think it is related to the End of Line characters. Would you be kind to download the latest version of bricklayers.py from the repository and try again?

Done, no difference.

Line number 2175 (+1 from last time, but presumably you made a one-line change earlier up).

GeekDetour, 2025-03-17:
Oh man.... that's so... infuriating 🤣 Thanks for your patience. I will include the checkings you did. But I hate to not understand what's really causing it. Thanks for your time on this!

jungletek, 2025-03-17:
You know, I'm not particularly handy with Python, C++ is more my jam... so I actually just asked Deepseek about the error message and showed it the script, it suggested the changes 😄 I know enough about programming to recognize a type error, but couldn't debug it myself.
I guess what I'm saying is that if you're stuck, it couldn't hurt to throw it at an AI and see if they can see anything obvious.

I know in the past when I've had to try to make Python scripts work cross-platform, that sometimes it was simply an issue of encapsulating a magic statement into quotes, or parentheses, etc.
Might be worth going over with some sort of linter/code-checker?

Dunno, I'm spitballing here. I'd be willing to step through the script in a debugger if that would help you, but I'm not really familiar with IDLE (for example), so you'd have to talk me through it and tell me where to watch in the code.

GeekDetour, 2025-03-17:
Jungletek: you got it and I appreciate it. Please, would you test the latest version?

GeekDetour, 2025-03-17:
You are right: this script already has 10 Classes... Each of them would become a separate file (Module, in Python)... BUT, it makes distribution of the script much less straightforward. A lot of people want just to download 1 file that does it. I don't want to create tooling to "pack" things together just for this purpose. 

I hope this thing doesn't grow much more. We all want this implemented in the slicers themselves. This code is a 'brainstorm' tool, in my view - for what we want (or don't want)

GeekDetour, 2025-03-18:
Oh, Jungletek - I forgot to tell you (and I just updated the README) - to use PYPY, it must be the **pypy3.exe** (not pypy.exe, because it most likely is Python 2).

jungletek, 2025-03-18:
> Oh, Jungletek - I forgot to tell you (and I just updated the README) - to use PYPY, it must be the **pypy3.exe** (not pypy.exe, because it most likely is Python 2).

Self-installed vanilla, system-wide-on-the-path Python 3.12 here, AND I also have it in my MSYS2 environment (and I have Linux machines I can test on too...)

GeekDetour, 2025-03-18:
Man, the time you dedicated today was amazing - I am so happy! I tested extensively on my Mac, also on my daughter's Windows 11 machine... and... it was running great! It is impossible to reproduce all the super different scenarios. You are helping so much. Thanks for the patience. But try PYPY! The difference in speed is surreal :) Man, enjoy! Print things! I read on TenTech's REPO someone saying that their prints were a bit stringy too... that might be something interesting to learn about Brick Layers... I am updating the Readme with anything I realize I was forgetting. Here in the issues, there will be people with simple doubts - I will be super happy if you drop by and find something you know how to help.

jungletek, 2025-03-18:
It would be nice if PrusaSlicer wouldn't ignore our system PATH variables. I can call python.exe, py.exe, etc. from anywhere on my command line because it's in my PATH variable, but PrusaSlicer doesn't care and requires you to explicitly specify the full path to the executable (which isn't what their docs would lead you to believe). It would also be nice if the error messages were better and more verbose/explicit.

FYI: CPython users who installed the py.exe component can just use "C:\Windows\py.exe" to invoke Python from the PP script dialog (or anywhere else, assuming they added it, or let the installer add it, to their PATH variable); it's much shorter and simpler. Might be worth it to add to the docs/readme.

GeekDetour, 2025-05-29:
@rodrykthingol and @jungletek made excellent observations about which settings should be disabled or enforced for this script to work fine - this is super nice! Thanks guys!

It's a shame I am not having more time right now to improve it and make everything work fine, regardless of the settings people use. The reality is: I need to chase money now, otherwise I cannot pay the bills 😬

Closing this ticket - but pinning it so people can find it easily.

Todo: extract the useful information and add it to the Readme

closed 2025-05-29 by GeekDetour

GeekDetour, 2025-05-30:
The original title of this issue was "Linux - PrusaSlicer bug" - I am changing it to "Settings that should be Avoided for now"