GeekDetour/BrickLayers #4: Prusaslicer Error 1

McRoth73, 2025-03-17:
I'm getting this error when I try to use this script on Windows 11.  I don't see much information on installing Pypy, I just extracted the files from the download to the directory in the error.  Is there something else I should have done?  I did add the pypy folder to my PATH.

Post-processing script C:\pypy3\pypy3.exe C:\Users\emera\OneDrive\Documents\BrickLayers-main\bricklayers.py -startAtLayer 3 -extrusionMultiplier 1.05 -enabled 1 on file C:\Users\emera\AppData\Local\Temp\.38940_0.gcode.pp failed.
Error code: 1

xpuipc, 2025-03-17:
Hi, thank you for this great script. I also tried this with latest (2.9.1) Prusa Slicer and "....\Programs\Python\Python313\python.exe". I get the following error:
Post-processing script C:\Users\*****\AppData\Local\Programs\Python\Python313\python.exe C:\Users\*****\OneDrive\#Share#\3D Printing\bricklayers.py -startAtLayer 3 -extrusionMultiplier 1.05 -enabled 1 on file C:\Users\spyy\AppData\Local\Temp\.22172_0.gcode.pp failed.
Error code: 2
I am using your sample stl => BrickLayersChallengeSimple.stl as "input"

I am on Windows 11.

GeekDetour, 2025-03-17:
McRoth73, could you try opening your CMD terminal on windows, and run the following (huge) line?

C:\pypy3\pypy3.exe C:\Users\emera\OneDrive\Documents\BrickLayers-main\bricklayers.py C:\Users\emera\OneDrive\Documents\BrickLayers-main\sample_gcode\Sample_BrickLayersChallengeSimple_5walls.gcode -outputFile C:\Users\emera\OneDrive\Documents\BrickLayers-main\sample_brick\test_brick.gcode -startAtLayer 3 -extrusionMultiplier 1.05 -enabled 1

If it gives you an error, copy the whole text for me, please? thank you!

rodrykthingol, 2025-03-17:
Hmm.. If that doesn't work try using this format:
"C:\Your\Path\To\Python\python.exe" "C:\Your\Path\To\Script\bricklayers.py" -startAtLayer 3 -extrusionMultiplier 1.05 -enabled 1;

xpuipc, 2025-03-17:
Hi, i think i found the problem. It is when Prusa slicer wants to export .bgcode.... it is compressed. If you "switch off" .bgcode in the slicer it works for me...

GeekDetour, 2025-03-17:
OH THATS VERY IMPORTANT! THANK YOU SO MUCH! YEAH. It cannot be binary.

xpuipc, 2025-03-17:
and....it has problems with spaces in the path like "/3D prints/bla/"

GeekDetour, 2025-03-17:
Well, it can have spaces (I have spaces where the files are in my mac), but you need to know how to escape them for your os :(

GeekDetour, 2025-03-17:
Someone still having problems?

McRoth73, 2025-03-17:
I did uncheck bgcode before.  I'll try again in a bit when I get my computer going.

McRoth73, 2025-03-17:
I'm still getting the error.  Your cmd line code did work though.  

C:\Users\emera>C:\pypy3\pypy3.exe C:\Users\emera\OneDrive\Documents\BrickLayers-main\bricklayers.py C:\Users\emera\OneDrive\Documents\BrickLayers-main\sample_gcode\Sample_BrickLayersChallengeSimple_5walls.gcode -outputFile C:\Users\emera\OneDrive\Documents\BrickLayers-main\sample_brick\test_brick.gcode -startAtLayer 3 -extrusionMultiplier 1.05 -enabled 1

▁▃▅▆ Brick Layers ▆▅▃▁  (v0.2.0-25-g90b4978)

 Input File:            C:\Users\emera\OneDrive\Documents\BrickLayers-main\sample_gcode\Sample_BrickLayersChallengeSimple_5walls.gcode
 Output File:           C:\Users\emera\OneDrive\Documents\BrickLayers-main\sample_brick\test_brick.gcode
 Extrusion Multiplier:  1.05
 Layer to Start:        3
 Layers to Ignore:      []
 Enabled:               1
 Verbosity:             2

[#############################################################################################################] 100.00%
Original Size: 316.9kB  Layers: 75  Lines: 9566

Execution time: 0.28 seconds

Prusa error: 
Post-processing script C:\pypy3\pypy3.exe C:\Users\emera\OneDrive\Documents\BrickLayers-main\bricklayers.py -startAtLayer 3 -extrusionMultiplier 1.05 -enabled 1 on file C:\Users\emera\AppData\Local\Temp\.9552_0.gcode.pp failed.
Error code: 1

GeekDetour, 2025-03-17:
1st: please get the latest version of the bricklayers.py again

2nd: on Printers, General, Firmware, you will need to uncheck the Support Binary G-Code
(I know it is sad, but it doesn't work with post-processing scripts).

Tell me how it goes!

McRoth73, 2025-03-17:
Unfortunately I had already gotten that version of bricklayers.py right before I tried this last time.  Also, that bgcode box is unchecked.
I'm curious about this part of the error though, C:\Users\emera\AppData\Local\Temp.9552_0.gcode.pp failed.  it is a gcode.pp that it's trying to process.  Is that correct?

GeekDetour, 2025-03-17:
Yes. That's the temporary file that the slicer generated, saved on disk, then it sends it as input-file for the post processing script. My script must do the changes and overwrites it. The script gets out and then the slicer finally saves it where you chose to export the file (or sends to the printer).

L-A-Sutherland, 2025-03-18:
> OH THATS VERY IMPORTANT! THANK YOU SO MUCH! YEAH. It cannot be binary.

you can add a check for binary to create a more useful error message.

`def gcode_opener(path, flags):
    file=os.open(path, flags)
    header=os.pread(file,4,0)
    if header==bytes("GCDE","ascii"):
        os.close(file) #do not leak the file descriptor
        print(f"{error_marker}The specified output folder '{input_file}' is a binary gcode file, which is not supported. Disable binary gcode in your slicer settings", file=sys.stderr)
        sys.exit(1)  # Exit immediately
    return file
`

`        with open(input_file, 'r', encoding="utf8", newline="",opener=gcode_opener) as infile, open(final_output_file, 'w') as outfile:
`

That should help people get it running.

McRoth73, 2025-03-18:
I just tried it again and the latest version of the .py works!  Thanks!

GeekDetour, 2025-03-18:
There you go @McRoth73 !! Great to know it is working for you. Closing this one.

closed 2025-03-18 by GeekDetour