GeekDetour/BrickLayers #10: Not sure if an issue - Bambu Slicer

Doyle4, 2025-03-18:
Downloaded latest build, got it all setup and once pressed 'Slice' a little python box appears with no errors.
Only thing thats got me wondering is if I have done something incorrect, once sliced I see no difference.
I have made a 'BrickLayers' profile with the command saved, once sliced I took a screenshot of a section of the sliced model and then sliced it without the bricklayers processing, took another screenshot and both look the same, normal sliced, no bricklayers visible.

Have I done something wrong?

Thanks

closed 2025-03-18 by Doyle4

Doyle4, 2025-03-18:
Its all working great thank you! Incredible work.
What made it easier for me was placing the Bricklayer.py file into the same folder as pypy3.exe.
I can simply swap that file once a new build has been uploaded.

Again, thank you for all this effort and making this free to everyone!

GeekDetour, 2025-03-19:
@Doyle4  Hi there! Yesterday some people helped a lot reporting problems on Windows - I don't have a Windows machine on my desk all the time, so very helpful to have more eyes! Looks like it reached an 'ok' level of stability now.

Have you printed something already? I am curious! Post photos if so - and what has been your impressions. Cheers!