GeekDetour/BrickLayers #11: Even amount of perimeters results in wrong offsets

flexbex, 2025-03-18:
When i have 4 perimeters i would expect perimeter 1,3 shifted together and 2, 4. But it is 1,4 together and 2,3

GeekDetour, 2025-03-18:
Please could you post a screenshot of? Thanks!!!

flexbex, 2025-03-18:
hope is good to see

[image]

flexbex, 2025-03-18:
try to get my head around it. When i understand correctly it is because the algorithm comes from inside and outside. So this would be a general problem for all walls consisting only perimeters with an even number

GeekDetour, 2025-03-18:
Let me give you an STL for you to slice and think about it... give me 5 minutes

GeekDetour, 2025-03-18:
Please, I just added this stl:
https://github.com/GeekDetour/BrickLayers/blob/main/sample_stl/cone_with_hole.stl

Slice it and you might understand the challenge (you don't want a sudden disturbance of the walls). Do a bricklayer thing on the part, and move the layers up and down.

flexbex, 2025-03-18:
ah it fails to open. But I think I get your point. Thank you anyway for your work. Love it. Will update if i find anything else.

closed 2025-03-18 by GeekDetour