GeekDetour/BrickLayers #12: Unicode Encode Error while running in the command line.

VovyCZ, 2025-03-18:
```
PS C:\3D\pypy3.11-v7.3.19-win64> C:\3D\pypy3.11-v7.3.19-win64\pypy3.exe C:\3D\Scripts\bricklayers.py -startAtLayer 3 -extrusionMultiplier 1.05 -enabled 1 C:\3D\STL\koncovka_PETG_12m47s.gcode

▁▃▅▆ Brick Layers ▆▅▃▁  (v0.2.0-37-ga835e16)

 Input File:            C:\3D\STL\koncovka_PETG_12m47s.gcode
 Output File:           C:\Users\zvavr\AppData\Local\Temp\tmp0lj0klms.gcode
 Extrusion Multiplier:  1.05
 Layer to Start:        3
 Layers to Ignore:      []
 Enabled:               1
 Verbosity:             2

Traceback (most recent call last):
  File "C:\3D\Scripts\bricklayers.py", line 2180, in <module>
    for processed_line in processed_gcode:
  File "C:\3D\Scripts\bricklayers.py", line 1506, in process_gcode
    bytes_received += len(line.encode("ascii"))
                          ^^^^^^^^^^^^^^^^^^^^
UnicodeEncodeError: 'ascii' codec can't encode character '\u0159' in position 23: ordinal not in range(128)
PS C:\3D\pypy3.11-v7.3.19-win64>
```

GeekDetour, 2025-03-18:
Can you provide me with the C:\3D\STL\koncovka_PETG_12m47s.gcode file please?

VovyCZ, 2025-03-18:
[koncovka_PETG_12m47s.zip](https://github.com/user-attachments/files/19325275/koncovka_PETG_12m47s.zip)

GeekDetour, 2025-03-18:
@VovyCZ you zipped me your PyPy folder :) I just wont the koncovka_PETG_12m47s.gcode if possible

VovyCZ, 2025-03-18:
> [@VovyCZ](https://github.com/VovyCZ) you zipped me your PyPy folder :) I just want the koncovka_PETG_12m47s.gcode if possible

I I have checked it, and the archive contains only one file: koncovka_PETG_12m47s.gcode.

GeekDetour, 2025-03-18:
Could you download the bricklayers.py script again (or git pull) and test in your machine?

VovyCZ, 2025-03-18:
> [@VovyCZ](https://github.com/VovyCZ) Could you download the bricklayers.py script again (or git pull) and test in your machine?

Now everything seems okay! Thank you so much!

VovyCZ, 2025-03-18:
> [@VovyCZ](https://github.com/VovyCZ) Don't go away! I think you are using Wall-Order Inner/Outer/Inner or Outer/Inner. Please change to Inner/Outer and check it out - you will have a much better brick-layers result.

I made the change. However, I still have a small issue; the text 'None0.0' appears on line 342 in the gcode file. I am attaching the updated gcode with the error."

[gcodes-2025218-203759.zip](https://github.com/user-attachments/files/19326578/gcodes-2025218-203759.zip)

GeekDetour, 2025-03-18:
@VovyCZ Thanks for pointing that out! I just fixed it. This is an enforcement of adding the ;WIDTH:xxx on every External perimeter, so in the Preview we have a correct visualization (otherwise, the Bricked Internal perimeters on the bricked layer can make the external perimiters that follows appear wrong in the preview. It doesn't really matter for when you print - but people expect the preview to look pristine. 
The problem is: The first time it ran, the right 'prefix' is not yet detected (None). These are the problems of identifying the right way to set ;WIDTH on every slicer. Bambu does in a different way than Orca and Prusa Slicer...
I have no idea if my explanation makes any sense, ha ha ha. You tell me.

Let me know if it fixed for you now! It should.

commit 599e4666 (2025-03-18): Fix UnicodeEncodeError when counting bytes in G-code files. Now using UTF-8 instead of ASCII. Closes #12.
closed 2025-03-19 by GeekDetour