GeekDetour/BrickLayers #15: module 'os' has no attribute 'pread' on Windows

adamus1red, 2025-03-18:
The `os.pread` added in 1fa24eedb73274cecda283d77c9de09d452d5ee4 breaks the script when running on Windows operating systems. 

```
PS C:\Users\A\Downloads> C:\Users\A\AppData\Local\Programs\Python\Python312\python.exe C:\Users\A\Downloads\3D\scripts\BrickLayers\bricklayers.py -startAtLayer 3 -extrusionMultiplier 1.05 -enabled 1 .\Box_Sc_PETG_6h50m.gcode

▁▃▅▆ Brick Layers ▆▅▃▁  (v0.2.0-51-g140e65b)

 Input File:            .\Box_Sc_PETG_6h50m.gcode
 Output File:           C:\Users\A\AppData\Local\Temp\tmpj0kikclv.gcode
 Extrusion Multiplier:  1.05
 Layer to Start:        3
 Layers to Ignore:      []
 Enabled:               1
 Verbosity:             2

Traceback (most recent call last):
  File "C:\Users\A\Downloads\3D\scripts\BrickLayers\bricklayers.py", line 2182, in <module>
    with open(input_file, 'r', encoding="utf8", newline="",opener=gcode_opener) as infile, open(final_output_file, 'w', encoding="utf-8", newline="\n") as outfile:
         ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\A\Downloads\3D\scripts\BrickLayers\bricklayers.py", line 2039, in gcode_opener
    header=os.pread(file,4,0)
           ^^^^^^^^
AttributeError: module 'os' has no attribute 'pread'. Did you mean: 'read'?
```

The same issue occurs when running the script with Bambu Studio

[image]

GeekDetour, 2025-03-19:
Good catch and I am sorry... fixing something and breaking something else, ha ha ha. Could you pull it and test it again? :)

adamus1red, 2025-03-19:
@GeekDetour Yeah, looks like it's now fixed in the latest version

```
PS C:\Users\A\Downloads\3D\scripts\BrickLayers> git pull
remote: Enumerating objects: 8, done.
remote: Counting objects: 100% (8/8), done.
remote: Compressing objects: 100% (4/4), done.
remote: Total 6 (delta 4), reused 4 (delta 2), pack-reused 0 (from 0)
Unpacking objects: 100% (6/6), 1008 bytes | 63.00 KiB/s, done.
From https://github.com/GeekDetour/BrickLayers
   4de12da..8f7279c  main       -> origin/main
Updating 4de12da..8f7279c
Fast-forward
 bricklayers.py | 15 +++++++--------
 1 file changed, 7 insertions(+), 8 deletions(-)
PS C:\Users\A\Downloads\3D\scripts\BrickLayers> cd ../../../
PS C:\Users\A\Downloads> C:\Users\A\AppData\Local\Programs\Python\Python312\python.exe C:\Users\A\Downloads\3D\scripts\BrickLayers\bricklayers.py -startAtLayer 3 -extrusionMultiplier 1.05 -enabled 1 .\Box_Sc_PETG_6h50m.gcode

▁▃▅▆ Brick Layers ▆▅▃▁  (v0.2.0-53-g757d5a0)

 Input File:            .\Box_Sc_PETG_6h50m.gcode
 Output File:           C:\Users\A\AppData\Local\Temp\tmpkdilewkv.gcode
 Extrusion Multiplier:  1.05
 Layer to Start:        3
 Layers to Ignore:      []
 Enabled:               1
 Verbosity:             2

[########################################################################################] 100.00%
Original Size: 19.4MB  Layers: 867  Lines: 809149

Peak memory usage: 49.83 MB
Execution time: 40.97 seconds
```

GeekDetour, 2025-03-19:
@adamus1red  GEEEZ... 40 seconds! 867 layers! Looks like some serious part you are testing there. By the way, try using PYPY - cutting time in 4

adamus1red, 2025-03-19:
> [@adamus1red](https://github.com/adamus1red) GEEEZ... 40 seconds! 867 layers! Looks like some serious part you are testing there. By the way, try using PYPY - cutting time in 4
TBH it shouldn't be since it's a fairly straightforward container. I'd guess it might be the ribs on the walls but I haven't dug into it.

[image]

Either way I think this issue has been resolved and can be closed.

commit 757d5a05 (2025-03-19): Fix Windows compatibility: Replace os.pread() with open().read() in gcode_opener()

- Replaced os.pread() with open().read() to ensure cross-platform support.
- Prevents 'module os has no attribute pread' error on Windows.
- Maintains same functionality while improving safety and readability.
- Automatically closes file handle before reopening the descriptor.

Fixes #15.
closed 2025-03-19 by adamus1red