GeekDetour/BrickLayers #5: Orcaslicer Error Code 1

HoneybadgerSi, 2025-03-18:
I'm receiving this error, I saw Prusaslicer has an option to disable binary g-code but I am not seeing this option in Orcaslicer. 

What options do I have to implement this post-processing script, or possible work arounds?

Thank you in advance. 

[image]

GeekDetour, 2025-03-18:
Just to be sure, can you paste here your "Post-Processing" scripts text?

GeekDetour, 2025-03-18:
"F:\3D Printing\OrcaSlicer\PyPy\pypy3.11-v7.3.19-win64\pypy3.11-v7.3.19-win64\pypy.exe" "F:\3D Printing\OrcaSlicer\PyPy\pypy3.11-v7.3.19-win64\pypy3.11-v7.3.19-win64\bricklayers.py" -startAtLayer 3 -extrusionMultiplier 1.05 -enabled 1

are you sure it woulnd't be: "F:\3D Printing\OrcaSlicer\PyPy\pypy3.11-v7.3.19-win64\pypy.exe" 
"F:\3D Printing\OrcaSlicer\PyPy\pypy3.11-v7.3.19-win64\bricklayers.py" 
-startAtLayer 3 -extrusionMultiplier 1.05 -enabled 1

Just checking - you are the only one to know where you created things.

GeekDetour, 2025-03-18:
Try changing for pypy3.exe

HoneybadgerSi, 2025-03-18:
It's not producing errors now with the recommended "pypy3.exe" and "-startAtLayer 3 -extrusionMultiplier 1.05 -enabled 1" 

However it doesn't seem to be creating the staggered layers (at least as viewed in preview) and crashes Orcaslicer when selecting more than 2 walls.

GeekDetour, 2025-03-18:
That's the thing we don't love about post-processing scripts. When you export the GCODE, the Preview doesn't automatically show the changes... You need to Drag the exported Gcode back to your slicer Window.

HoneybadgerSi, 2025-03-18:
It seems to be working then, I'll see if I can workaround the issue with "4 walls" next then. 

Thanks for the help, first time ever touching python.

GeekDetour, 2025-03-18:
Oh, try disabling ARC.

HoneybadgerSi, 2025-03-18:
I already had it disabled. 

I'm going to go through and check more settings (I'll start by just running presets with the script) then add those settings back until something breaks again)

GeekDetour, 2025-03-18:
Please, if you can mark as closed, awesome!

closed 2025-03-18 by HoneybadgerSi