GeekDetour/BrickLayers #6: Staggered Inner Seams crashes Orcaslicer with Bricklayers.

HoneybadgerSi, 2025-03-18:
Only setting so far I've found that breaks the script [in OrcaSlicer at least].

rodrykthingol, 2025-03-18:
By what i've gathered the crucial settings are activating wipe on retraction, deactivate scarf joint, activate inner/outer and activate detect thin lines. These seem to be the most optimal settings, at least on prusa slicer. 

Though your mileage may vary.

GeekDetour, 2025-03-18:
@HoneybadgerSi and @rodrykthingol 

Thank you so much for your effort, guys. I'll try my best to make this script play well with all the possible settings to be useful - some might require some refactoring. Your investigations to which options are breaking it... are such a big help! I appreciate the time into this.

The crazy thing is: one day the major slicers will implement Brick Layers natively. So, this script WILL be history. I say this just to emphasize **we are working on a program that will die**. When? I don't know. But keep that perspective - try to identify the problems that affect you right now, for your personal usage. But don't go crazy on the amount of hours dedicated to this.

This is a weird message, isn't it? "Please, help! But not so much.... ha ha ha"

GeekDetour, 2025-03-19:
Hi @HoneybadgerSi Are we good for now? Please, if you discover any identifiable problem, please, would you open a new ticked for the specific details? I'll do my best to fix it.

closed 2025-03-19 by GeekDetour

HoneybadgerSi, 2025-03-19:
Thanks, got busy! 

Looking forward to this, I have a lot of high strength requiring parts and projects so this kind of feature package is going to becoming a pillar of our process.

GeekDetour, 2025-03-19:
@HoneybadgerSi Do you perform any measurement / breaking tests? I am extremely curious to see more people testing it and discussing their findings.