GeekDetour/BrickLayers #7: bricklayers

alphasteve706, 2025-03-18:
been trying for a few hors now and i keep getting this error what could i be doing wrong

Post-processing script "C:\Users\user1\AppData\Local\Microsoft\WindowsApps\python.exe" "C:\Users\user1\Downloads\bricklayers.py" -startAtLayer 3 -extrusionMultiplier 1.05 -enabled 1 on file C:\Users\user1\AppData\Local\Temp\.7044_0.gcode.pp failed.
Error code: 9009

alphasteve706, 2025-03-18:
and i have moved my folder for differnt paths with no luck and unchecked bgcode box im using windows 11

GeekDetour, 2025-03-18:
Just to check... Is it Python3 that you have installed?

alphasteve706, 2025-03-18:
How do I check im new to this

alphasteve706, 2025-03-18:
I downloaded from your link then I typed in python. Where python and copied path

GeekDetour, 2025-03-18:
Open your terminal (cmd) window and C:\Users\user1\AppData\Local\Microsoft\WindowsApps\python.exe -V

alphasteve706, 2025-03-18:
K I will that's like command prompt

alphasteve706, 2025-03-18:
Python not found download from Microsoft store or disable this short cut

alphasteve706, 2025-03-18:
C:\Users\user1>C:\Users\user1\AppData\Local\Microsoft\WindowsApps\python.exe -V
Python was not found; run without arguments to install from the Microsoft Store, or disable this shortcut from Settings > Manage App Execution Aliases.\

GeekDetour, 2025-03-18:
try:
where python

alphasteve706, 2025-03-18:
Microsoft Windows [Version 10.0.22631.4890]
(c) Microsoft Corporation. All rights reserved.

C:\Users\user1>where python
C:\Users\user1\AppData\Local\Microsoft\WindowsApps\python.exe

rodrykthingol, 2025-03-18:
Mmm.. First off. Try uninstalling python and reinstalling it.

For windows: 

https://www.geeksforgeeks.org/how-to-install-python-on-windows/

Make sure that you check the ADD python to PATH.

[image]

Then go to your command line and do where python and just copy paste

"C:\Your\Path\To\Python\python.exe" "C:\Your\Path\To\Script\bricklayers.py" -startAtLayer 3 -extrusionMultiplier 1.05 -enabled 1;

On PrusaSlicer, make sure that you have something similar to this. 

[image]

If you encounter more issues, check:
https://github.com/GeekDetour/BrickLayers/issues/2

There are couple of tips and tweaks there.

alphasteve706, 2025-03-18:
I reinstall python and now it works but only without your extrusion multiplier line when I paste that I get a different error I will post it here but I am getting brick layers with im very excited about because I do alot of functional prints.. so that you man

GeekDetour, 2025-03-18:
@alphasteve706 please, paste here your "Post-processing scripts" section as you configured in your slicer. Let me give a fresh look at it :)

alphasteve706, 2025-03-18:
Gotcha

alphasteve706, 2025-03-18:
t

alphasteve706, 2025-03-18:
with this"C:\Users\user1\AppData\Local\Programs\Python\Python313\python.exe" "C:\Users\user1\Documents\birick layers\bricklayers.py"; -startAtLayer 3 -extrusionMultiplier 1.05 -enabled 1;

 command line i get this error The configured post-processing script does not exist: -startAtLayer

GeekDetour, 2025-03-18:
There is a ";" before the -startAtLayer that should be removed 🤗

alphasteve706, 2025-03-18:
> @alphasteve706 <https://github.com/alphasteve706> There is a ";" before

Oh ha ha k

alphasteve706, 2025-03-18:
Have you ever thought about a getting a discord server going

jungletek, 2025-03-18:
> Have you ever thought about a getting a discord server going

Right tool for the job, and all that...

Why would Discord be better than Github for tracking issues?

GeekDetour, 2025-03-19:
@alphasteve706 Is it working for you now? Do you close it? Or I do :)

closed 2025-03-19 by GeekDetour