GeekDetour/BrickLayers #8: Invalid speed in 'G1 E-0.00 F0 ; BRICK: Retraction'

flexbex, 2025-03-18:
Prusaslicer and klipper

I get the retraction and unretract error because "g1 ... f0" I guess. My settings in Prusaslicer are these

[image]

[image]

flexbex, 2025-03-18:
Manually replacing all f0 with the original values works

[probe_holder-lid_gcode.txt](https://github.com/user-attachments/files/19307044/probe_holder-lid_gcode.txt)

L-A-Sutherland, 2025-03-18:
I'll look into this. Can you post a gcode file that hasn't been put through the script?

flexbex, 2025-03-18:
sure

[unmodified_gcode.txt](https://github.com/user-attachments/files/19308464/unmodified_gcode.txt)

[prusa_config_ini.txt](https://github.com/user-attachments/files/19308473/prusa_config_ini.txt)

L-A-Sutherland, 2025-03-18:
can you enable wipe into infill and see if that works?
Looks like the script doesn't know how to react when there is no wipe at all.

L-A-Sutherland, 2025-03-18:
I modified the script replacing 

```     else:
            # wipe with travel:
            gcodes.append(from_gcode(f"G1 E-{stopped_pull:.2f} F{int(simulator.retraction_speed)} ; BRICK: Retraction \n"))
            gcodes.append(from_gcode(feature.const_wipe_start))
            gcodes.append(from_gcode(f"G1 F{int(simulator.wipe_speed)}\n"))
            for point, extrusion in zip(moving_points, moving_extrusions):
                gcodes.append(from_gcode(f"G1 X{point.x:.3f} Y{point.y:.3f} E-{extrusion:.5f} ; BRICK: Wipe \n"))
            gcodes.append(from_gcode(feature.const_wipe_end))
            gcodes.append(from_gcode(f"G1 X{target_state.x} Y{target_state.y}{hopping_z} F{int(simulator.travel_speed)} ; BRICK: Target Position\n"))
            if z is not None:
                gcodes.append(from_gcode(f"G1 Z{z:.2f} ; BRICK: Target Position\n"))
            gcodes.append(from_gcode(f"G1 E{simulator.retraction_length:.2f} F{int(simulator.retraction_speed)} ; BRICK: Urnetract\n"))
```
with
```
elif simulator.retraction_speed>0 and simulator.wipe_speed>0:
            # wipe with travel:
            gcodes.append(from_gcode(f"G1 E-{stopped_pull:.2f} F{int(simulator.retraction_speed)} ; BRICK: Retraction \n"))
            gcodes.append(from_gcode(feature.const_wipe_start))
            gcodes.append(from_gcode(f"G1 F{int(simulator.wipe_speed)}\n"))
            for point, extrusion in zip(moving_points, moving_extrusions):
                gcodes.append(from_gcode(f"G1 X{point.x:.3f} Y{point.y:.3f} E-{extrusion:.5f} ; BRICK: Wipe \n"))
            gcodes.append(from_gcode(feature.const_wipe_end))
            gcodes.append(from_gcode(f"G1 X{target_state.x} Y{target_state.y}{hopping_z} F{int(simulator.travel_speed)} ; BRICK: Target Position\n"))
            if z is not None:
                gcodes.append(from_gcode(f"G1 Z{z:.2f} ; BRICK: Target Position\n"))
            gcodes.append(from_gcode(f"G1 E{simulator.retraction_length:.2f} F{int(simulator.retraction_speed)} ; BRICK: Urnetract\n"))
        else:
            gcodes.append(from_gcode(f"G1 X{target_state.x} Y{target_state.y}{hopping_z} F{int(simulator.travel_speed)} ; BRICK: Target Position\n"))
            if z is not None:
                gcodes.append(from_gcode(f"G1 Z{z:.2f} ; BRICK: Target Position\n"))`
```
That seemed to cause the gcode output to be valid with no wipes.

GeekDetour, 2025-03-18:
Yeah, blind spot of mine. Would you make a pull request?

L-A-Sutherland, 2025-03-18:
Can do.

GeekDetour, 2025-03-18:
It was merged. I'll need to revisit my code, because you can have retractions without the wiping and right now I think it is not being done unless Wiping was activated - as soon as I am rested.

GeekDetour, 2025-03-18:
By the way, Sutherland, thanks for caring. I can see you were also following the TenTechs implementation of it. I am curious about your take on the whole thing, what you expect, and if you printed something already using bricklayers. I appreciate it.

GeekDetour, 2025-03-18:
@flexbex I am closing this one. I think it was resolved for you right? Reopen please if needed!

closed 2025-03-18 by GeekDetour

L-A-Sutherland, 2025-03-18:
@GeekDetour I forked TenTech's brick layers repo to improve the zigzag generation described in "vertical brick layers". I also modified that script to do wide lines instead of zigzags. Those are the only forms of brick layer that i have printed so far, and they seem promising. I find the whole brick layer concept to be fascinating. 

For your script I've not printed anything yet, but judging by the gcode previews it should work well. In my own branch I've been looking at the build_loop_tree function. I've made a version of it that works bidirectionally so that you don't need to merge the lists afterwards. May be able to solve the OrcaSlicer out of order problem with it too.

GeekDetour, 2025-03-18:
@L-A-Sutherland That's so cool man! I am happy that my code is... understandable? It's been 20 years since I was actually paid to do Python stuff. I am incredibly out of shape and probably out of tune to the way people are coding today. I apologize for crazy style quirks. 

About the out of order problem: I have a major flaw in my script: I am am currently doing the bounding box detection and the build_loop_tree on the loops after an internal perimeter finishes... It is actually premature. The only absolute way to make it bullet proof would be to do it with ALL the LOOPS of the current layer (or at least after an object is completed and it changes to the next object). So, there should me a major refactoring in gathering those inner perimeter loops on a data structure that would allow a complete analiysis and - after it is decided what goes brick and what stays, the perimeters that stay should be inserted on their corresponding points on the gcode. Maybe you might see the problem in a simpler and more effective way.

Tell me what you think.

Once more, I really appreciate your brain on this.