GeekDetour/BrickLayers #18: Seam Issue

Doyle4, 2025-03-19:
I did a 30x30 4wall Zero infill to test.
I found with the Bricklayer process, when it comes to the starting point its over extruding (i think) or the K-Values are ignored?

I disabled the Bricklayer command in Bambu Studio and reprinted at 20x20 and the issue was gone as expected.

Any ideas what could be causing this?

[image]
[image]

GeekDetour, 2025-03-19:
This is just 2 walls, without any infill, is it? 

Thanks giving it a try and posting the photo, it is a type of print that I usually don't do.

Can you try it with 4 walls?

PS: I am not working on this test right now - I will assign it to me when I'll be able to.

GeekDetour, 2025-03-19:
@Doyle4  Oh, do me a favor? Could you just include here your BambuStudio project (the Cube), so I might play with your settings?

Doyle4, 2025-03-19:
@GeekDetour Here you go 👍 

I do have settings override in the filament settings, without these I have print quality issues, thought Id give you the headsup so you can look in there also.

Thanks

GeekDetour, 2025-03-19:
@Doyle4 I just put a Warning in the README of the Repo about print quality. Something went bad along the way. I am focusing on this now. It wasn't like this at all.

GeekDetour, 2025-03-19:
I am making some major refactoring on the Wipe / Retraction / Travel functions. Hopefully it will get Brick Layers printing better than before - and the code easier to understand and improve. Thanks everybody. Let's see if I can do it in one day or two.

pathurus, 2025-03-19:
🤞🏻

Doyle4, 2025-03-20:
@GeekDetour Thank you for looking into this, look forward to future updates.
Great to see so many interested.

dmjohn0x, 2025-03-20:
Ive gotten the same results printing This articulated Trumptopus toy (https://www.printables.com/model/1194866-trumptopus-articulated-octopus) to test if tolerances were effected, and im noticing much poorer quality on the articulated arms vs the head. Ive reprinted this 4 times. Then two with the bricklayering script have noticeably worse quality on the legs using the same print settings and filament (Polymaker Panchroma CoPE).

GeekDetour, 2025-03-20:
@dmjohn0x Any part the depends heavily on retractions, my feeling is that BrickLayers is not a good fit. Specially small parts. It is very prone to stringing. I hope we dial in things well. This is VERY experimental so far. Which printer are you using?

Doyle4, 2025-03-20:
Tested on Windows with latest BrickLayers.py and sadly its far worse than before.
I have attached an screenshot of the imported G-code in Bambu Studio.
30x30 cube, 3 bottom layers, 4 walls 0 top layers

[image]

GeekDetour, 2025-03-20:
I am sorry @Doyle4 ... there is a limit to what I can guess by not being at your machine.... I understand your are disappointed / not more than me, believe 😳 It visualizes correctly, I am printing parts.... I hope someone with a Windows machine can help us

VovyCZ, 2025-03-20:
"Yes, something is very wrong in the latest version. I tried to slice a Voron cube and this is how it turned out."

[image]

VovyCZ, 2025-03-20:
> [@VovyCZ](https://github.com/VovyCZ) I think you might be using Arachne in OrcaSlicer. Please, try the Classic Engine. It's in the documentation.

No, I have it set correctly. Orca 2.3.0, Windows 11, bricklayers.py v0.2.1-1-gbb10780.

Doyle4, 2025-03-20:
@GeekDetour Going to check now if the latest is doing the same with OSX for me.

SeductiveFox, 2025-03-20:
Had a similar issue with a benchy I tried to print. It doesn't look quite that bad lol but definitely got a lot of stringing and looks like some retraction issues overall maybe. 
[image]
[image]

GeekDetour, 2025-03-20:
@SeductiveFox the latest version I just pushed fixed yet some bugs related to retractions.

Anyways, there is a possibility that Brick Layers is prone to stringing. Or at least we would need to understand more about the process to fine tune it.

SeductiveFox, 2025-03-21:
> [@SeductiveFox](https://github.com/SeductiveFox) the latest version I just pushed fixed yet some bugs related to retractions.
> 
> Anyways, there is a possibility that Brick Layers is prone to stringing. Or at least we would need to understand more about the process to fine tune it.

Awesome! I'll have to try it out and see how it goes! Its completely understandable that there will be issues, considering how new this all is!

GeekDetour, 2025-03-21:
@Doyle4 Thanks a lot for your tests and eyes :) I was absolutely confident that it was perfect. The Benchy was perfect on Prusa Slicer, my little test piece with overhangs was looking perfect as well - it printed well too. AND then when you guys posted your screenshots... WOW! And it was all true: still some logic mistakes.

Doyle4, 2025-03-21:
@GeekDetour Not a problem :) Happy to help where I can, I apologise I can't help more than only test builds that are released.

Unfortunately, the latest build I am still having the same issues as before, once layer 9 is complete it'll skip to top layer.
I have attached a OSX exported g-code, its also the same on Windows. 
I have added the log file also incase you can see anything in there :)

[OSX-Current-BLpy.gcode.3mf.zip](https://github.com/user-attachments/files/19383407/OSX-Current-BLpy.gcode.3mf.zip)

[bricklayers_log.txt.zip](https://github.com/user-attachments/files/19383417/bricklayers_log.txt.zip)

GeekDetour, 2025-03-21:
@VovyCZ That makes me SO HAPPY! Thank you! Post photos please! 
It is not perfect yet, but the latest update made me believe that it could potentially become a tool people use for real, not just a proof of concept.

Doyle4, 2025-03-21:
@GeekDetour I found something interesting, using the latest build I managed to slice a pea shooter fine (not yet tested a print) but the BrickLayer.py seems to be having issues with a single cube. I created one in Bambu Studio adding a primitive and it fails, I made a cube using Tinkercad with same dimensions and same slicing settings and had the same result as a Primitive cube.

[Latest Tests.zip](https://github.com/user-attachments/files/19395770/Latest.Tests.zip)
Both of these files where sliced with version v0.2.1-8-gc6a5f41.

GeekDetour, 2025-03-21:
Yeah, I spent some time with the files you sent but I couldn't figure out yet. I am more interested in the settings you used. Could you save the Project? 3MF?? ❤️

VovyCZ, 2025-03-21:
> [@VovyCZ](https://github.com/VovyCZ) That makes me SO HAPPY! Thank you! Post photos please! It is not perfect yet, but the latest update made me believe that it could potentially become a tool people use for real, not just a proof of concept.

[image]
[image]
[image]

Doyle4, 2025-03-21:
@GeekDetour 
Yeah its a strange one.
Added my settings.
[Cube.3mf.zip](https://github.com/user-attachments/files/19397016/Cube.3mf.zip)

Doyle4, 2025-03-22:
Seems the cube is only thing struggling at the moment, other models Iv tested are fine.

Wonder if it's to do with zero infill?

VovyCZ, 2025-03-22:
@Doyle4 

When I open your 3mf file in Orca, I get this error message and the cube cannot be sliced.

Error:
"Relative extruder addressing requires resetting the extruder position at each layer to prevent loss of floating point accuracy. Add "G92 E0" to layer_gcode."

Doyle4, 2025-03-22:
@GeekDetour Could try these, and export them and see if the gcode does the same?

GeekDetour, 2025-03-22:
I am probably detecting something bad that is creating code that would print correctly, but the preview doesn't like it / and hide the lines. I am not sure if it is the layers with different height.
Also, it seems that when you generate gcode for Bambu Printers, the layers get numbered / and I am creating a new intermediate layer for the "bricked" (shifted) loops - and I am not numbering them.
So, I need to come up with a way to reprocess the numbered layers (that, by the way, include the total number of layers - which in my case is still a problem because by design I don't load the whole gcode in memory, I process line by line, and write down all the lines I reprocess when there is a layer chance). These were design choices for being able to process huge GCode files, even if the machine doesn't have a lot of RAM. So, I need to re-evaluate if those decisions are good ones or bad ones.

This is the problem of parsing and changing a GCode, which is very different than what the slicer does: it is creating the code once, with all the logic it has in place. With this script we are trying to modify the GCode substantially, and generate a changed one that, not only would print well but also works well with the Slicer Preview 🥵❤️🤣

Doyle4, 2025-03-22:
@GeekDetour I really appreciate your work on this and explaining your process throughout.
Again, I do apologise I cant help on the coding side as its not my strong point, Id just mess it up a lot more :( haha 
But for early stages, this is fantastic.

dmjohn0x, 2025-03-25:
> [@dmjohn0x](https://github.com/dmjohn0x) Any part the depends heavily on retractions, my feeling is that BrickLayers is not a good fit. Specially small parts. It is very prone to stringing. I hope we dial in things well. This is VERY experimental so far. Which printer are you using?

Bambu Labs P1S w/ a .04 Nozzle using Orca Slicer 2.3.0-dev. I'll update the script and look to see if anything is different.

dmjohn0x, 2025-03-26:
@GeekDetour , The latest version of the script has really screwed up surface layers. Im not sure how the script works so its difficult to explain exactly what Im talking about so I'll link a quick picture: 

[image]

Kind of looks like severe under extrustion... but the issue was fixed by removing the post processing script. I tried multiple models and the surface was messed up on all of them after updating the bricklaying script last night.

GeekDetour, 2025-03-26:
@dmjohn0x Hi there! Would you be kind to attach your project? 3MF or the STL that you are printing? Or at least the GCode that was exported.

Indeed the models I tested are mostly vertical things / not so much flat.

I would like to analise it.

Thanks!

dmjohn0x, 2025-03-26:
> [@dmjohn0x](https://github.com/dmjohn0x) Hi there! Would you be kind to attach your project? 3MF or the STL that you are printing? Or at least the GCode that was exported.
> 
> Indeed the models I tested are mostly vertical things / not so much flat.
> 
> I would like to analise it.
> 
> Thanks!

Sure. Heres the gcode(I had to zip the file, as github wont let me share gcode files.): 

[Cylinder_PETG_1h4m.zip](https://github.com/user-attachments/files/19476405/Cylinder_PETG_1h4m.zip)

GeekDetour, 2025-03-29:
@dmjohn0x You have a very peculiar sense of humor 😂 🤷‍♂️

I think you know the holes have nothing to do with Brick Layers: it is a top surface with very little support underneath. It will print poorly in any printer.

[image]

GeekDetour, 2025-03-29:
I will close this issue. The very bad quality problem was fixed. Now we are perfecting retraction / travel / wiping trying to eliminate stringing.

closed 2025-03-29 by GeekDetour