GeekDetour/BrickLayers #19: Orcaslicer Error code 1

ShinnAshian, 2025-03-19:
"C:\3DPrinting\pypy3.11-v7.3.19-win64\pypy3.exe" "C:\3DPrinting\pypy3.11-v7.3.19-win64\bricklayersA.py" -startAtLayer 3 -extrusionMultiplier 1.05 -enabled 1;

I'm trying [v0.2.0-53-g757d5a0](https://github.com/GeekDetour/BrickLayers/commit/8f7279c984c5e1c8cb60b937faddf9d8d29e4d24) on [orcaslicer 2.30](https://github.com/SoftFever/OrcaSlicer/releases/tag/v2.3.0-rc) currently. 

I've already read the previous issue about this error code but the solution seemed not working on my setup. Instead it just keeps popping this error code. There was a black terminal window or something pop up blinkly when I tried to save gcode.
[image]

GeekDetour, 2025-03-19:
Please, would you be kind to change to -enabled 0 (disables in), exporting your GCODE file again, and post it here?

ShinnAshian, 2025-03-19:
Thanks for attention!

It's popping error code 2 instaed of 1 now. When I cleared texts in the post processing script slot it succesfully saved a gcode. https://1drv.ms/u/c/3ad5b20144ef75a6/EYoNvE1kYU5CvTXt8IfZ-P0B2aFD26VlydT1WXj5YBdrNQ?e=DJjSzc
([image])

I guess it might be some kind of problems with the portable pypy. gonna try fresh install a python to see if it helps.

> [@ShinnAshian](https://github.com/ShinnAshian) Please, would you be kind to change to -enabled 0 (disables it), export your GCODE file again, and post it here? So I can try 'bricking' it here and see whats going on.

GeekDetour, 2025-03-19:
Can we make another test?
- copy your Test5_PETG_1h20m.gcode to inside C:\3DPrinting
- Open your windows terminal (cmd, or PowerShell
- Copy the following line to run:

C:\3DPrinting\pypy3.11-v7.3.19-win64\pypy3.exe C:\3DPrinting\pypy3.11-v7.3.19-win64\bricklayersA.py C:\3DPrinting\Test5_PETG_1h20m.gcode -outputFilePostfix _brick -startAtLayer 3 -extrusionMultiplier 1.05

If it gives you any error, I would need the full text.
Or, tell me if it ran nice (it would generate another file: _brick.gcode, if it runs nice)

ShinnAshian, 2025-03-19:
C:\3DPrinting>C:\3DPrinting\pypy3.11-v7.3.19-win64\pypy3.exe C:\3DPrinting\pypy3.11-v7.3.19-win64\bricklayersA.py
  File "C:\3DPrinting\pypy3.11-v7.3.19-win64\bricklayersA.py", line 7
    <!DOCTYPE html>
     ^
SyntaxError: invalid character '!' (U+0021)

[image]

As I deleted the character '!',  <title>BrickLayers/bricklayers.py at main 路 GeekDetour/BrickLayers</title> showed up.

[image]

Which is diffrent from what it is in the original file <title>BrickLayers/bricklayers.py at main · GeekDetour/BrickLayers</title>

I consulted LLM and it said it's the python interpreter encounters unrecognised characters while parsing code. And there's only chinese language pack installed on my windows PC. I'll install the english language pack and feed back.
_btw that's a random model i randomly pulled into orca for testing the script only, thanks for the advice!_

> I am asking you to test because I could run your gcode perfectly here. But I need to say the model you chose is not a good test for brick layers: it has an almost horizontal overhang near the borders. If you actually try to print it, the borders will droop a lot.

GeekDetour, 2025-03-19:
@ShinnAshian it looks like you didn't actually downloaded the python script, instead it is an HTML page. Did you download the actual repository as a Zip file (or cloned the repository)?

closed 2025-03-19 by GeekDetour

ShinnAshian, 2025-03-20:
> [@ShinnAshian](https://github.com/ShinnAshian) it looks like you didn't actually download the python script, instead it looks like an HTML page. Did you download the actual repository as a Zip file (or cloned the repository)? Be sure to at least press the "raw" button that downloads the python script.

My bad, it's working right now and thanks for your greatwork!

GeekDetour, 2025-03-20:
@ShinnAshian That's awesome! :)