GeekDetour/BrickLayers #20: OrcaSlicer or Bambu Studio not working with X1C

TexturedGuppy, 2025-03-19:
Not generating the bricklayer layers with script on X1C printer. 

Have this pasted into Post Processing Scripts on both Orca and Bambu:

`"C:\Users\Cameron\AppData\Local\Programs\Python\Python313\python.exe" "K:\3D printer models\BrickLayersGeekDetour\BrickLayers-main\bricklayers.py" -startAtLayer 2 -extrusionMultiplier 1.05 -enabled 1;`

Can see window pop up and script runs for a second in background, and it is successfully slicing.

When inspecting each layer inside and outside walls are equal heights. 

Have classic wall generator set as well as walls set to Inner/Outer.

[image]

[image]

[image]

GeekDetour, 2025-03-19:
Hi @TexturedGuppy - it is sad, but NONE of the Slicers update the Preview to show the results of Post-Processing Scripts automatically. We need to drag the exported GCODE back to the Slicer to actually see it.

Export to disk, drag the file back to the slicer and you will see the result.

On BambuSlicer it is a bit hidden: you need to go to File > Export > Export G-code to do that.

TexturedGuppy, 2025-03-19:
Oh thanks I had no clue, that did the trick!

closed 2025-03-19 by GeekDetour

null-except, 2025-03-23:
Maybe the script could have a small GUI widget after the slicing with a message like "all done, export .gcode and open in a new instance to see the changes"?

GeekDetour, 2025-03-23:
@null-except I don't know if the slicers give any information for the script for us to differentiate when we are a) exporting gcode to the disk VS b) sending directly to the printer... So far, for the post-processing scripts, it's all the same.

You got me curious - when I have time I will check if there is any clue for us to use. But I believe we have the hands tied about this one 😐

Anyways, two thoughts about it: if we ever mature the script enough for us just trusting "sending directly to the printer without ever checking", then it won't matter. And second: I do hope slicers implement Brick Layers eventually, and this script will be forgotten 🤣

L-A-Sutherland, 2025-03-25:
@GeekDetour We can tell if it's being sent directly to the printer using an environment variable set by the slicer: "SLIC3R_PP_HOST"
It is set to "File" when you're saving to the disk. Not sure what exact value it has when exporting directly to the printer but it should be different.

GeekDetour, 2025-03-25:
@L-A-Sutherland Interesting! TODO: check if the environment variable is different in every slicer, ha ha ha :) Maybe printing out all of them and see what they are: 

import os
for name, value in os.environ.items():
    print("{0}: {1}".format(name, value))

Torino460, 2025-05-11:
If we don't generate gcode.  Will it still print in bricklayers if I just go ahead and send it after I slice it with the scripts?  Mine wont send the saved gcode to my bambu x1c for some reason.  Oh and I tried to use my ankermake and it did not work.  It started to print but wasn't extruding or even close to the plate.