GeekDetour/BrickLayers #23: Add simple curl command to README for easier installation

null-except, 2025-03-22:
Hi! 👋

This project is really awesome – thanks for sharing it!

To make it easier for end users who might be new to GitHub, it could be helpful to include a one-liner in the README for installing (and updating) the script, like this:

`curl -O https://raw.githubusercontent.com/GeekDetour/BrickLayers/main/bricklayers.py`

It’s simple, works across platforms, and the same command can be used for both installation and updates.

Apologies if this suggestion isn’t relevant, just thought it might help a few newcomers. 😊

Thanks again!

GeekDetour, 2025-03-22:
This is a great suggestion. I'll put it in my list to create an easy to install tutorial.

andor-pierdelacabeza, 2025-04-03:
Maybe "compiling" the Python script in a binary that includes pypy/CPython could be nice so people don't have to mess up with their terminal...

@GeekDetour , do you have any docs on testing/development/benchmark to see if any packaging or modification to the code changes the functionality ?

fangxx3863, 2025-04-12:
@GeekDetour I suggest you try to compile it as a separate binary distribution using nuitka, which can be done with a simple command  
`nuitka --mingw64 --standalone --show-progress --show-memory --onefile bricklayers.py`  
It will convert Python into C code and compile it into binary, which will bring considerable performance improvements.  

CPython  
[image]

Nuitka  
[image]  

Here is a built Windows binary for testing  
[windows-bricklayers.zip](https://github.com/user-attachments/files/19721738/windows-bricklayers.zip)

GeekDetour, 2025-04-13:
@fangxx3863 That's very interesting! I didn't know Nuitka.

It might be cool in the future for generating a 'binary' version of bricklayers (once it is a bit more polished). It seems it is harder to provide debugging information.

About performance, PyPy is still faster. Here is my results, in my machine, comparing normal Python, PyPy and a compiled version with Nuitka:

- CPython:   1.52s  
- PyPy:   0.46s  
- Nuitka Compiled:  1.09 seconds

PyPy is still my recommended option for people who want Python Scripts to run fast - it just runs, no need to compile every time you make changes to the script.

fangxx3863, 2025-04-15:
> [@fangxx3863](https://github.com/fangxx3863) That's very interesting! I didn't know Nuitka.
>
> It might be cool in the future for generating a 'binary' version of bricklayers (once it is a bit more [...]

You are absolutely right. However, for those who are not familiar with the command line, being able to compile it into a standalone binary file is more important than performance improvement (it can be used right out of the box).