GeekDetour/BrickLayers #24: Make it work with Variable Layer Height

notnotluna, 2025-03-23:
The readme doesn't state if it does or not, I don't think it should be too hard to implement if it isn't already

GeekDetour, 2025-03-23:
It should already 🤗 I read the layer height at every layer and create an intermediate layer half that height up. 👍 That's why you don't need to pass the layer height in the calling of the script.

null-except, 2025-03-24:
It does work! My first try with BrickLayer was with Variable Layer Height! 

[image]

The only downside is that the gcode preview is a bit weird when you're only looking at one layer at a time.

GeekDetour, 2025-03-24:
@null-except that's great!!! Could you print it?? THAT's what I am really curious to see! Thanks for testing!!!!

null-except, 2025-03-24:
Yes, I have already printed some pieces, no defect on my parts, was really just perfect!

GeekDetour, 2025-03-29:
It looks like it works fine with Adaptive Layers. I am closing this one 👍

closed 2025-03-29 by GeekDetour