GeekDetour/BrickLayers #27: Bambu labs error

JoeKstar, 2025-03-28:
Put post processing script as described in instructions i press slice get error 2?

Doyle4, 2025-03-28:
What operating system are you using?
Also, screenshots help a lot :)

ToolandTrade, 2025-03-30:
I've tried installing this and using it on Bambu Studio as well, but I'm getting error messages. I'm a newb to all this so I'll just detail exactly what I did.

Downloaded and extracted PyPy3 to its own folder in Program Files (x86)
Downloaded and extracted Bricklayers folder within the PyPy3 Folder to its own folder
Imported .stl file to print in Bambu Studio
Turned off Arc Fitting
Made sure Walls are set to print "Inner-Outer"
Copied paths to PyPy3 and Bricklayers files as follows (without quotation marks):
C:\Program Files (x86)\pypy3.11-v7.3.19-win64\pypy3.11-v7.3.19-win64\pypy3.11.exe;
C:\Program Files (x86)\pypy3.11-v7.3.19-win64\BrickLayers-main\BrickLayers-main\bricklayers.py
Each time I try to slice the plate, I get an error code. The first time I got
Screenshot 2025-03-30 083146

I tried starting a new project (repeating the above steps) and a different PyPy3 .exe file (thinking maybe I linked to the wrong one) and I'm now getting
Screenshot 2025-03-30 084756

I tried installing Python 3 instead and repeating the above steps and keep getting similar error messages.

I'd really love to try Brick Layers to see how it might improve the strength of my prints, so any help is much appreciated!

jungletek, 2025-03-31:
> [image]

Judging by the way Bambu Studio thinks the script is at `C:\Program` (which likely doesn't exist, and is the wrong path anyway), you DO need to encapsulate your path(s) with quotes.

For example, for me in PrusaSlicer, the line to call the script looks like `"C:\Python\Python312\python.exe" "C:\Users\USERNAME\Documents\Python Scripts\BrickLayers\bricklayers.py" -startAtLayer 3 -extrusionMultiplier 1.05 -enabled 1;`

Notice that I have to encapsulate the paths to Python and the BrickLayers script, because it has spaces in it. If your path(s) have spaces, you'll likely have to do the same because it seems that Bambu Studio isn't handling it for at least the script location, if not Python.

ToolandTrade, 2025-03-31:
> > [image]
> 
> Judging by the way Bambu Studio thinks the script is at `C:\Program` (which likely doesn't exist, and is the wrong path anyway), you DO need to encapsulate your path(s) with quotes.
> 
> For example, for me in PrusaSlicer, the line to call the script looks like `"C:\Python\Python312\python.exe" "C:\Users\USERNAME\Documents\Python Scripts\BrickLayers\bricklayers.py" -startAtLayer 3 -extrusionMultiplier 1.05 -enabled 1;`
> 
> Notice that I have to encapsulate the paths to Python and the BrickLayers script, because it has spaces in it. If your path(s) have spaces, you'll likely have to do the same because it seems that Bambu Studio isn't handling it for at least the script location, if not Python.
> 
> [@GeekDetour](https://github.com/GeekDetour) You should perhaps consider editing the Windows syntax examples in the readme to use (double)quote-encapsulated paths, or even create (or solicit) examples from each slicer on each platform, so you can know which one can handle paths with spaces, and which need them encapsulated. I can provide images from PrusaSlicer and Cura as examples on Windows if you need.

HECK YES! Thank you so much, it's working now!

ToolandTrade, 2025-03-31:
This is the form that worked for me on Bambu Studio BTW 

"C:\Program Files\Python313\python.exe" "C:\3D Printing\BrickLayers-main\BrickLayers-main\bricklayers.py" -startAtLayer 3 -extrusionMultiplier 1.05 -enabled 1;

GeekDetour, 2025-04-01:
@jungletek Thanks for suggesting the quotes around the path, in the Windows Examples. I updated the README ❤️👍

closed 2025-04-01 by GeekDetour