GeekDetour/BrickLayers #28: Lines on top of ironed layer

spiderixax, 2025-03-28:
With ironing enabled, some lines appear on top of the ironed top layer, see the screenshot below.

[image]

Doyle4, 2025-03-28:
I can imagine in the future you'll be able to set when to stop bricklayers so the top layers aren't affected.
Bricklayers with Ironing will be a mess, Ironing is one layer been melted by the nozzle with a small amount of filament to fill the gaps... bricklayers of course will make this worse as it uses two sets of layers for one layer.

GeekDetour, 2025-03-29:
@spiderixax Yes, so far I am not making any treatment to identify the "top" surface. It is on my list of desired features.

null-except, 2025-03-29:
You can simply add a -ignoreLayers option right?

GeekDetour, 2025-03-29:
@null-except yes, you could do that. But you need to do it manually for each model you parse. It would be much better to identify when a top-surface happens and not make the brick-layer thing on those areas. But again... I really hope Slicers might implement this whole intelligence natively and this script would rest in peace, ha ha ha. Of course, before that happens, I'll be improving this thing on the areas that are important to me and to people using it :)

GeekDetour, 2025-04-01:
@L-A-Sutherland That's a good idea, Luke. I'll pin this issue for reference.

closed 2025-04-01 by GeekDetour