GeekDetour/BrickLayers #30: Weird console error

ssmb212, 2025-04-01:
Hi geekdetour, i have been following and watching your videos for a longtime. I remember when you first spoke about brick layers and i was like doesnt someone just do it. Well here we are today, great thanks and its a great feature.  After slicing in orca slicer and exporting the gcode allseems fine and checking the gcode the layers different height is there. 

While printing though i am getting errors like that on klipper console at fluidd 

19:21:33 
// EXCLUDE_OBJECT_END NAME=ASSEMBLY_ID_0_COPY_0 does not match the current object NAME=ASSEMBLY_ID_1_COPY_0
19:21:41 
// Unknown command:"E0.00000"
19:21:41 
// Unknown command:"E0.00000"

Does this mean it doesnt recognize/use the height changes is it something i did wrong or should i just ignore it ?

GeekDetour, 2025-04-01:
@ssmb212 hi there! Could you export the GCODE and post it here? I'll need to take some time to look into it. Thanks for trying the script and for reaching out.

ssmb212, 2025-04-02:
Thank you for looking at it, heres the gcode. 

Seems i cant upload gcode type file here. Had to use external link.

https://drive.google.com/file/d/1lTdqwP-4zw853vfnV-2Ott1NUS2Q-43c/view?usp=drive_link

A heads up, initially i had 3 walls set up but the model was getting sliced ending up with 2 walls. After i increased the walls to 5 it ended up getting sliced to a 3 walls gcode. After the printing and strength testing it the inside was as follows. 2 Walls, gap 3rd wall and then infill.  Let me know if you can figure anything out.

ssmb212, 2025-04-10:
Oh my god i missed the email that google drive was asking me to confirm the file share so sorry, and i was wondering why you havent replied xD.

ssmb212, 2025-04-18:
> [@ssmb212](https://github.com/ssmb212) hi there! Could you export the GCODE and post it here? I'll need to take some time to look into it. Thanks for trying the script and for reaching out.

any luck ?

jungletek, 2025-04-21:
> Seems i cant upload gcode type file here. Had to use external link.

For reference, you could always just add an extra .txt extension to the end of "filename.GCODE" and the upload form treats it like a text file, no external hosting needed.

ssmb212, 2025-04-21:
> > Seems i cant upload gcode type file here. Had to use external link.
> 
> For reference, you could always just add an extra .txt extension to the end of "filename.GCODE" and the upload form treats it like a text file, no external hosting needed.

Thanks for the info, here i is again. 

[4.gcode.txt](https://github.com/user-attachments/files/19837439/4.gcode.txt)

ssmb212, 2025-05-17:
This was never checked :(