GeekDetour/BrickLayers #34: MCU 'mcu' shutdown: Stepper too far in past and Move exceeds maximum extrusion (284.976mm*2 vs 7.000mm*2) See the 'max_extrude_cross_section'

CodeZ1LLa, 2025-04-08:
I ran into the issue when the printer switches to the second layer. 

It throws:

Error message
{"code":"key112", "msg": "Move exceeds maximum extrusion (284.976mm*2 vs 7.000mm*2) See the 'max_extrude_cross_section' config option for details", "values": [284.976, 7.000]}

I increased the max_extrude_cross_section, but it still won't print. 

Printer: Ender 3
Runs on Klipper
Prusa Slicer

Hundreds of prints without brick layers were printed errorless, so it is definitely not printer-related. 
The problem is with printer.cfg or some settings in Prusa. 
Binarry G-Code is disabled, and the wall order is set from inner to outer ("External perimeters first" is disabled). 
I have a heavily modified config in Prusa, so something else may be causing it. 

I'd appreciate any help!

jungletek, 2025-04-08:
I believe I'm encountering the same/similar error. The "MCU shutdown" stuff is Klipper-specific, the real indicator of what the issue is is the reference to "maximum extrusion". IIRC this is due to the different extrusion modes (relative/absolute) being used, and the state not properly being accounted for. On other setups this might be fine, but I believe Klipper keeps track/calculates how much is/will be extruded, and if it seems to be too much (because the extrusion mode state wasn't handled properly), it throws this max. extrusion error and cancels the print.

Here's what my Klipper console output looks like, so you can see the similar message:
```
12:20 PM
Move exceeds maximum extrusion (37.834mm^2 vs 0.640mm^2)
12:20 PM
Move exceeds maximum extrusion (37.834mm^2 vs 0.640mm^2)
See the 'max_extrude_cross_section' config option for details
12:20 PM
EXCLUDE_OBJECT_END NAME=TESTBODY1_STL_ID_0_COPY_0 does not match the current object NAME=None
12:20 PM
EXCLUDE_OBJECT_END NAME=TESTBODY1_STL_ID_0_COPY_0 does not match the current object NAME=None
12:19 PM
EXCLUDE_OBJECT_END NAME=TESTBODY1_STL_ID_0_COPY_0 does not match the current object NAME=None
12:19 PM
EXCLUDE_OBJECT_END NAME=TESTBODY1_STL_ID_0_COPY_0 does not match the current object NAME=None
12:16 PM
Mesh Bed Leveling Complete
12:16 PM
probe at 172.337,126.462 is z=3.397500
12:15 PM
probe at 149.988,126.462 is z=3.502500
12:15 PM
probe at 127.650,126.462 is z=3.555000
12:15 PM
probe at 127.650,109.538 is z=3.570000
12:15 PM
probe at 149.988,109.538 is z=3.507500
12:15 PM
probe at 172.337,109.538 is z=3.415000
12:15 PM
probe at 172.337,92.600 is z=3.402500
12:15 PM
probe at 149.988,92.600 is z=3.497500
12:15 PM
probe at 127.650,92.600 is z=3.552500
12:15 PM
Adapted probe count: (3,3)
12:15 PM
Ratio: (0.2483, 0.183135135135)
12:15 PM
Adapted mesh bounds: ([87.653, 86.302],[132.347, 120.182])
12:15 PM
Original probe count: (10,10)
12:15 PM
Original mesh bounds: ((30.0, 20.0),(210.0, 205.0))
12:15 PM
Found 1 objects
12:10 PM
File selected
```

I've attached the STL I was printing, and the bricked GCode (remove the .txt extensions). According to Klipper, the print failed during layer 8, so that seems a sensible place to see if extrusion mode switches are being handled properly. Additionally, the `EXCLUDE_OBJECT_END NAME=TESTBODY1_STL_ID_0_COPY_0 does not match the current object NAME=None` errors occurring right before (note the timestamps), suggests to me that there might be some correlation.

[TestBody1.stl.txt](https://github.com/user-attachments/files/19653224/TestBody1.stl.txt)
[TestBody1_35m_0.24mm_210C_PLA_ENDER3V2.gcode.txt](https://github.com/user-attachments/files/19653223/TestBody1_35m_0.24mm_210C_PLA_ENDER3V2.gcode.txt)

CodeZ1LLa, 2025-04-08:
The issue was resolved. 
The cause is in setting "Use relative E distances" that was unchecked in my case. 
After turning relative E distances **on**, all my old settings, including auto layer thickness, work perfectly fine and it prints as it should

[image]

closed 2025-04-08 by CodeZ1LLa

jungletek, 2025-04-09:
Took a nap and it's resolved when I wake up. Nice! 😄
Makes sense, I was just too tired to test that at the time.

Can confirm that my setting in PrusaSlicer was the same (it's the default), and that enabling it solves this (at least it seems to, the print I printed before that gave this error made it to layer 16 this time before bed adhesion failed, lol. I've just moved over to PrusaSlicer from Cura when you released your script, so it looks like there's still a bit more tuning I need to do to my base profile(s).)

@GeekDetour : So, it seems like the only change you need to make is to point out that people need to enable relative extrusion distances in their slicer as part of the setup for the script.

GeekDetour, 2025-04-15:
@jungletek In theory, the script would work fine with both Relative and Absolute extrusions… What I didn’t test was actually printing with Klipper. So, yes, using Relative Extrusions is safer (much simpler). It is still in my todo list check what is going on with absolute - I thought the math was working correctly. But… yeah. Use Relative for now 🤣

jungletek, 2025-04-21:
Understood, happy to see you're dedicated to making this as robust as possible. Respect.