GeekDetour/BrickLayers #38: Could this be the solution for multi material?

L0rdS474n, 2025-04-21:
Just played a little with 3DBenchy in Bambu Lab Studio, include a ZIP with 2 versions of 3DBenchy. One single filament and one multi filament version, both with 10 wall thicknes just to have something to test with.
Include the full bricklayers.py file too.

Changes are made in parse_gcode_line in both of them.

[BrickLayers.zip](https://github.com/user-attachments/files/19834544/BrickLayers.zip)

L0rdS474n, 2025-04-21:
It needs to have exception for support materials, right now it just mix them in so it gets weird.
But it do slice well even for H2D with materials on different tools.

GeekDetour, 2025-04-21:
@L0rdS474n Looks like you just added this: 
line = line.replace(',', '.')

I am curious to understand why replacing the Commas into Dots Help with Multi-Filament 🤔

L0rdS474n, 2025-04-21:
As soon as I add more materials I get this error

ValueError: could not convert string to float: '0,500000'

So maybe not a universal fix but at my system it will fix the issue

Med vänliga hälsningar

Henrik Holmberg

PGP: https://keybase.io/d313373_m3/pgp_keys.asc
<https://keybase.io/d313373_m3/pgp_keys.asc?fingerprint=6d2458b57310f9744398b2168924d5b5da251eb8>

GeekDetour, 2025-04-22:
@L0rdS474n Which Slicer and Printer are you using? Yesterday I was checking the GCode again and float numbers were all using . for decimals (not , ) - so I was really intrigued.

GeekDetour, 2025-04-22:
@L0rdS474n interesting! I found reports of some users in European Countries in which slicers - due to regional settings in Windows, decimal numbers were being written with , instead of periods. The problem is: the Official GCode support in most firmwares is expecting periods.
Bambu Studio had an issue in version 1.9.0.60 (which version are you using?)

L0rdS474n, 2025-04-22:
Latest version they have released

Med vänliga hälsningar

Henrik Holmberg

PGP: https://keybase.io/d313373_m3/pgp_keys.asc
<https://keybase.io/d313373_m3/pgp_keys.asc?fingerprint=6d2458b57310f9744398b2168924d5b5da251eb8>