GeekDetour/BrickLayers #39: gives an error in orca

OXOTHUK1978, 2025-04-27:
[image]

OXOTHUK1978, 2025-04-27:
what am I doing wrong

GeekDetour, 2025-04-29:
@OXOTHUK1978  Those Paths must be really correct to what is in your machine, both the path to the Python.exe as well as where you saved your bricklayers.py

OXOTHUK1978, 2025-04-29:
bricklayers.py i copy the folder and write the path to it. I also specify the path to the installed python. I've already tried different ways, sometimes error 2, sometimes error 5

OXOTHUK1978, 2025-04-29:
[image]

[image]

OXOTHUK1978, 2025-04-29:
thank you!!! figured it out

namewithnumber, 2025-04-30:
I'm encountering the same issue, but in a slightly different way. I got mine worked out last night before I went to bed, but somehow this Error popped up this morning. I haven't changed any settings, and swapped from Python to pypy, the same message keeps popping up. 
I copied and pasted the path using the 'Copy by Path' tool; I don't think a path error is the cause. 

But whenever I export a G-code file. It either gives me Error code 2 or Error code 1. 

If I use 

"E:\Python\pypy3.11-v7.3.19-win64\pypy3.exe"
"E:\Test\BrickLayers-main\bricklayers.py"-startAtLayer 3 -extrusionMultiplier 1.05 -enabled 1;

It will give me error code 1,

If I use 

"E:\Python\pypy3.11-v7.3.19-win64\pypy3.exe""E:\Test\BrickLayers-main\bricklayers.py"-startAtLayer 3 -extrusionMultiplier 1.05 -enabled 1;

It will give me error code 2. 

And if try 

"E:\Python\pypy3.11-v7.3.19-win64\pypy3.exe""E:\Test\BrickLayers-main\bricklayers.py"
-startAtLayer 3 -extrusionMultiplier 1.05 -enabled 1;

It will give me "The configured post-processing script does not exist: -startAtLayer"

A different layout may result in another kind of Error. The only thing I know is that it worked once and not anymore. 

[image]

jungletek, 2025-05-01:
It took me a minute to spot it, but you're omitting spaces in both instances.

In the first instance, there's no space between your last quotation mark and the arguments you're trying to pass to the script. So add a space between the `"` and `-startAtlayer`.

In the second instance, you have the same issue, but you're also missing a space between the path to python, and the path to the script. That will never work as the system will just treat it like one long string of the 2 paths smashed together.

The third instance won't work because adding a carriage return makes the program think you want to execute 2 different scripts, the latter of which doesn't exist (which it tells you).

TL;DR: You had it working, then you managed to delete a critical space and didn't notice. Technically it didn't "stop working", you just broke it and didn't notice it. Pretty common mistake.

Also for future reference, if you put console commands and output inside a code block like I did in my third sentence,, it's significantly easier to spot stuff like this (for me at least) because of the fixed-width font.

namewithnumber, 2025-05-01:
> Also for future reference, if you put console commands and output inside a code block like I did in my third sentence, it's significantly easier to spot stuff like this (for me at least) because of the fixed-width font.

Thanks, man, I was so confused about this error. LOL, the script thing is very new to me. Now I know how to get it to work.

jungletek, 2025-05-03:
> thank you!!! figured it out

You should "close" the issue if it's resolved, FYI.

> Thanks, man, I was so confused about this error. LOL, the script thing is very new to me. Now I know how to get it to work.

You're very welcome, glad I could help you understand a little better. Have fun!

GeekDetour, 2025-05-29:
@jungletek Thank you very much for helping people out to run the script properly - it's not straightforward when people are doing it for the first time 🤣

I'll mark this one as closed.